easyeda/easyeda-std-i18n #3: Comment not in English

fabianoriccardi, 2021-07-11:
What does it mean the following comment, found [here](https://github.com/EasyEDA2021/easyeda-i18n/blob/f59c17f967a5bb0508baa3eacd4a6e35663b0f4c/editor-std/ui_data_it.txt#L1930)?

    #后端返回英文的错误信息

EasyEDA2021, 2021-07-12:
Hi
The "#" charater start line, just the comment line, please ingore it

closed 2021-07-12 by EasyEDA2021

fabianoriccardi, 2021-07-12:
I know what `#` means. However they can help to understand how the file is organized.

EasyEDA2021, 2021-07-12:
Hi
If you want to know what it means, please translate it at Google translation, we will remove it at next time.
the organization these is no specific organization, the editor will find the text if it is match the string, if the string has duplication, the editor will decide to get which one to translate, 
for example:
`
#price
[price]
free = 免费
[/price]

#freedom
[freedom]
free = 自由
[/freedom]
`