easyeda/easyeda-std-i18n #54: Unexpected lines in "Layers and objetcts" window in PCB editor for Portuguese

GabPGomes, 2021-09-12:
Hi
I've downloaded the latest version (6.4.24) of easyEDA and have found something a bit odd. Some lines (as in one of the pictures), had the suffix ".layer". I expected it to disappear when added to the software's UI but that didn't happen (as you can see in the other picture). Should I change anything or leave it as it is?
[image]
[image]

Thank you

EasyEDA2021, 2021-09-13:
Hi
You don't need to think about the dot charater, just translate this whole string.

You can check Chinese translation ui_data_hans.txt

EasyEDA2021, 2021-09-13:
ALL.layer = 全部层
Copper.layer = 铜箔层
Non-Copper.layer = 非铜箔层
Other.layer = 其他层

GabPGomes, 2021-09-18:
Problem solved!

closed 2021-09-18 by GabPGomes