easyeda/easyeda-std-i18n #63: There are possible English spelling in English file

paulsteigel, 2021-09-17:
I found some terms of clarity, possible spelling mistakes:

Line 3172: should be Placement Tolerance rather than
Plcarment Tolerance 

Line 3172:
(less then 0.15mm), 
should be (less than 0.15mm)
Similarly applied to line 3133: (less then 0.08mm) >> (less than 0.08mm)

In several cases "can not less " seems to be incorrect, should be |can not be less " >> a replace all should be ok! Similarly applied for "can not" >> "cannot"
Line 2439: "charaters" should be characters
Line 2290: "must larger" >> should be "must be larger"
Line 2266: "must more" >> should be "must be more"
Line 3135: "System footprint must exist pad number"  >> should be "System footprint must have pad number"
Line 2057: "thirdparty" >> should be "third-party"
Line 1314: "Found the complicated polygon object can not" >> should be "Found the complicated polygon that object cannot"
Some more I have not recalled yet!
Please check with thanks!
Dang Dinh Ngoc
PS. Minor but good to do some checks

EasyEDA2021, 2021-09-18:
Yes, you are right, some sentense is not good, we will update it, thank you

paulsteigel, 2021-09-18:
Can you publish the Vietnamese translations. I can check from the interface to verify the accuracy of context based translation. Vietnamese is a difficult language and not as precise as English. It must be based on the using context.
Thank you!

EasyEDA2021, 2021-09-26:
> 
> 
> Can you publish the Vietnamese translations. I can check from the interface to verify the accuracy of context based translation. Vietnamese is a difficult language and not as precise as English. It must be based on the using context.
> Thank you!

we will update it at next release

fabianoriccardi, 2021-11-03:
[image]

In the top menu, Should it be "Settings" instead of "Setting"?

fabianoriccardi, 2021-11-06:
Moreover:

- line 1892: Operation --> Operations
- line 2733: Member --> Members

EasyEDA2021, 2021-12-24:
we changed these words, close the topic

closed 2021-12-24 by EasyEDA2021