easyeda/easyeda-std-i18n #72: Need feedback on translation method

EasyG0ing1, 2021-09-25:
Hello,

I spent the last day or so writing a Java program that engages the Google Cloud API for language translation and from my initial experiments, it seems to do pretty well with your document. However, I did notice some areas that would be better to translate by breaking the line down into different parts and translating those words then re-assembling the line so that some of the non-language portions of the document can remain properly placed in the translated side of the document.

Would you look at some of these translations that I did and id you can provide me with any kind of rules or text patterns to look for so that I can guarantee that they remain properly placed in the translation side ... I think it might be possible to translate into every language in a manner of minutes once the program is properly handling the areas of the document that won't translate straight across.

Anyways, I look forward to your feedback.

Mik
[Translations.zip](https://github.com/EasyEDA2021/easyeda-i18n/files/7230153/Translations.zip)
e

EasyEDA2021, 2021-09-26:
Hi
Thanks for your hard work, 
we have been think about this method by using Google translation API, the translation system not good enough than people translation, so we recommended translate the file by people.

thanks

EasyEDA2021, 2022-01-17:
Hi
we release v6.4.29, you can check it now

thank you

closed 2022-01-17 by EasyEDA2021

EasyG0ing1, 2022-01-17:
My local install of EasyEDA hasn't picked up on an upgrade yet ... I'm sure it takes time after you push it maybe?

EasyEDA2021, 2022-01-17:
The editor v6.4.29 translation is contain at this pull require
https://github.com/EasyEDA2021/easyeda-i18n/pull/101