easyeda/easyeda-std-i18n #74: Found a few improper lines in template, need help clarifying

EasyG0ing1, 2021-09-26:
Hello, I found these lines in the template that both start with # symbol and also have an = at the end.

Can you tell me which of these lines should have the # symbol removed and which ones should have the = sign removed so I can fix the template and issue a pull request?

Also, is it OK to remove the first space after any line starts with a # symbol?  Look at line numbers 2529 and 2530 for example ... would it be OK to remove the space after the # symbol and would that be universally ok to do throughout the entire template?

```
Line Number,Line Text
263,#Terms of Use & Privacy Policy =
271,#PCB Fabrication Capabilities =
963,#Duplicated document title =
986,#Clone it to a project =
987,#Clone it, then edit it =
1019,#It will take a while to unroute this net, do you want to continue? =
2300,#Ripup Nets =
2529,# Column Spacing =
2530,# Row Spacing =
3022,Assign LCSC Part# =
3079,# Rename Folder =

```

Thank you,

Mike

EasyEDA2021, 2021-09-27:
hi
all lines start with # are commet lines, or they are not using now.
LCSC Part# is a normal string

thanks

EasyG0ing1, 2021-09-29:
Thank you, I fixed these lines in the CLEAN template and created a Pull Request.

closed 2021-09-29 by EasyG0ing1